ethz-asl/voxblox #343: Build Standalone Without ROS

apache8080, 2020-10-21:
As the title suggests, I am trying to use voxblox without ROS and I was wondering if I could build it without ROS. It seems like core voxblox relies on catkin in the CMakeLists and gets eigen from ROS. Could I get away with just writing a new CMakeLists.txt or will I also have to make code changes?

alexmillane, 2020-10-21:
There are two things being conflated here. There is ROS and catkin. The voxblox core library does not use ROS features, but it is built under catkin.

You could try to build the package outside catkin, but currently we link against other packages also built in catkin which are defined in the package.xml file:

```  
  <depend>eigen_catkin</depend>
  <depend>eigen_checks</depend>
  <depend>gflags_catkin</depend>
  <depend>glog_catkin</depend>
  <depend>minkindr</depend>
  <depend>protobuf_catkin</depend>
```

So unless you want to try to build those packages outside of catkin too, you're gonna need a catkin workspace, in which case you should just build voxblox there.

Is there a super good reason for not building in catkin, other than it seeming a bit pointless if you're not using ROS features?

apache8080, 2020-10-21:
So I tried just installing catkin on a machine and using catkin to build but it doesn't work due to orocos_kdl being a plain cmake file. I may be doing something wrong. Here are the steps I followed:
```