ethz-asl/voxblox #366: undefined references to google::protobuf but building protobuf_catkin just fine

luisfer12, 2021-05-07:
I am building protobuf_catkin just fine but voxplot doesn't seem to be linked correctly. I have seem many similar issues online but no suggestion seems to fix my problem. Any suggestion of what I could try?

```
Errors     << voxblox:make /home/luisfer/catkin_ws/logs/voxblox/build.make.000.log                                                                                                                         
/home/luisfer/catkin_ws/devel/lib/libvoxblox_proto.so: undefined reference to `google::protobuf::RepeatedField<unsigned int>::unsafe_data() const'
/home/luisfer/catkin_ws/devel/lib/libvoxblox.so: undefined reference to `google::protobuf::io::CodedOutputStream::WriteVarint32SlowPath(unsigned int)'
/home/luisfer/catkin_ws/devel/lib/libvoxblox.so: undefined reference to `google::protobuf::io::CodedOutputStream::CodedOutputStream(google::protobuf::io::ZeroCopyOutputStream*)'
/home/luisfer/catkin_ws/devel/lib/libvoxblox_proto.so: undefined reference to `google::protobuf::internal::ArenaImpl::AllocateAligned(unsigned long)'
/home/luisfer/catkin_ws/devel/lib/libvoxblox_proto.so: undefined reference to `google::protobuf::internal::WireFormat::SerializeUnknownFieldsToArray(google::protobuf::UnknownFieldSet const&, unsigned char*)'
/home/luisfer/catkin_ws/devel/lib/libvoxblox_proto.so: undefined reference to `google::protobuf::internal::WireFormat::SerializeUnknownFields(google::protobuf::UnknownFieldSet const&, google::protobuf::io::CodedOutputStream*)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/luisfer/catkin_ws/devel/lib/voxblox/test_load_esdf] Error 1
make[1]: *** [CMakeFiles/test_load_esdf.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/luisfer/catkin_ws/devel/lib/libvoxblox_proto.so: undefined reference to `google::protobuf::RepeatedField<unsigned int>::unsafe_data() const'
/home/luisfer/catkin_ws/devel/lib/libvoxblox.so: undefined reference to `google::protobuf::io::CodedOutputStream::WriteVarint32SlowPath(unsigned int)'
/home/luisfer/catkin_ws/devel/lib/libvoxblox.so: undefined reference to `google::protobuf::io::CodedOutputStream::CodedOutputStream(google::protobuf::io::ZeroCopyOutputStream*)'
/home/luisfer/catkin_ws/devel/lib/libvoxblox_proto.so: undefined reference to `google::protobuf::internal::ArenaImpl::AllocateAligned(unsigned long)'
/home/luisfer/catkin_ws/devel/lib/libvoxblox_proto.so: undefined reference to `google::protobuf::internal::WireFormat::SerializeUnknownFieldsToArray(google::protobuf::UnknownFieldSet const&, unsigned char*)'
/home/luisfer/catkin_ws/devel/lib/libvoxblox_proto.so: undefined reference to `google::protobuf::internal::WireFormat::SerializeUnknownFields(google::protobuf::UnknownFieldSet const&, google::protobuf::io::CodedOutputStream*)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/luisfer/catkin_ws/devel/lib/voxblox/tsdf_to_esdf] Error 1
make[1]: *** [CMakeFiles/tsdf_to_esdf.dir/all] Error 2
make: *** [all] Error 2
cd /home/luisfer/catkin_ws/build/voxblox; catkin build --get-env voxblox | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

```

myboyhood, 2025-01-15:
@luisfer12 hello, I meet the same problem, but I have solved it!! 
 I provide a possible solution as follows:
 use this comand in CmakeLists.txt
```c
message(STATUS "protobuf_catkin_INCLUDE_DIRS: ${protobuf_catkin_INCLUDE_DIRS}")
message(STATUS "protobuf_catkin_LIBRARY_DIRS: ${protobuf_catkin_LIBRARY_DIRS}")
message(STATUS "protobuf_catkin_LIBRARIES: ${protobuf_catkin_LIBRARIES}")
```

In my case, I find 
```c
[cmake] -- protobuf_catkin_INCLUDE_DIRS: /usr/include
[cmake] -- protobuf_catkin_LIBRARY_DIRS: 
[cmake] -- protobuf_catkin_LIBRARIES: /usr/local/lib/libprotobuf.a;-lpthread
```
But I find the static library still throw errors like `undefined reference to xxxx`

So I try to use dynamic library explicityly
use `dpkg -L libprotobuf-dev` to find your dynamic library, in my case , it is located in `/usr/lib/x86_64-linux-gnu/libprotobuf.so`

So change the CmakeLists.txt like follows:
```c
set(CMAKE_FIND_LIBRARY_SUFFIXES .so)  # 
set(PROTOBUF_USE_STATIC_LIBS OFF)    # 
link_directories(/usr/lib/x86_64-linux-gnu)

find_package(Protobuf REQUIRED)
message(STATUS "PROTOBUF_LIBRARIES: ${PROTOBUF_LIBRARIES}")

include_directories(${Protobuf_INCLUDE_DIRS})

# the most important is add dynamic lib explicityly
target_link_libraries(${PROJECT_NAME}_proto
  /usr/lib/x86_64-linux-gnu/libprotobuf.so
  -lpthread
)
```

myboyhood, 2025-01-15:
Also you should comment this `protobuf_catkin` package in `voxblox_https.rosinstall`
```sh
#- git:
#    local-name: protobuf_catkin
#    uri: https://github.com/ethz-asl/protobuf_catkin.git
```

Then clean your workspace, execute this again
```sh
wstool init . ./voxblox/voxblox_ssh.rosinstall
wstool update
```

myboyhood, 2025-01-15:
## Another findings!
If you encounter the protobuf version conflict, 
```sh
[libprotobuf FATAL ../third_party/protobuf/src/google/protobuf/stubs/common.cc:87] This program was compiled against version 3.0.0 of the Protocol Buffer runtime library, which is not compatible with the installed version (3.13.0).  Contact the program author for an update.  If you compiled the program yourself, make sure that your headers are from the same version of Protocol Buffers as your link-time library.  (Version verification failed in "/home/zph/ros_ws/voxblox_ws/build/voxblox/compiled_proto/voxblox/Block.pb.cc".)
```

I find the voxblox link my static libaray  `/usr/local/lib/libprotobuf.a`, but I have explicitly linked my dynamic library `/usr/lib/x86_64-linux-gnu/libprotobuf.so`. So they make conflict.

**My solution** is `sudo mv libprotobuf.a libprotobuf.a.back` temporaly, and `catkin build voxblox --verbose`