ethz-asl/voxblox #372: Missing parameter "output_mesh_as_pointcloud"

r-judeh, 2021-08-05:
I am trying to publish the generated mesh as a point cloud on the topic "mesh_pointcloud" by setting the parameter "output_mesh_as_pointcloud" to true as explained in the documentation. However, this is not working and after looking at the source code, it seems that there is nothing in the "tsdf_server.cc" that allows this to happen. I am not sure if this is just a mistake in the documentation or the implementation is actually there but I cannot find it. So, could you please elaborate more on this, and if the functionality is not there, could you give a suggestion on how to add it? Thank you

magnusschmidt, 2022-03-21:
try adding this to your launch file
`<param name="publish_pointclouds" value="true" />`

salihmarangoz, 2022-03-22:
https://github.com/ethz-asl/voxblox/pull/168