ethz-asl/voxblox #373: Artifacts on Reconstruction (KITTI)

nachovizzo, 2021-09-10:
I'm running Voxblox as a baseline for some experiments and I found out some artifacts on the reconstruction.

I'm directly using the C++ API. In the past, I did use the ROS wrappers, that apparently get rid of some part of these artifacts:
[image]

By doing some experiments (decreasing voxel size) I realize that the "hole" that appears in the ground plane looks to be coming from a wrongly aligned point cloud:
[image]

Do you know what could be the cause? I'm not used to the "minkidir" library, therefore I can't really backtrack bugs there.

Thanks a lot!!!!

Kin-Zhang, 2022-11-04:
the default config in ROS wrap launch, enable_icp=true

and about the align one, I think we must use the `minkidir` for voxblox which has a totally different inverse log etc, basic operator, compared with Eigen. check here:
https://github.com/ethz-asl/minkindr/blob/master/minkindr/include/kindr/minimal/implementation/rotation-quaternion-inl.h

Leave for afterward experiments.

nachovizzo, 2022-11-04:
@Kin-Zhang Would you like to contribute the fix using minkidir to the pybind repo?