ethz-asl/voxblox #386: How to get esdf map from tsdf map using voxblox?

Lbaron980810, 2021-11-25:
Hi, there. Thanks for making voxblox and voxblox_ground_truth open source, both of them are great works and helpful!
Recently I'm trying to get a esdf map of gazebo world, so I used voxblox_ground_truth to get the tsdf file first, then used voxblox to translate tsdf map to esdf map. However, there's nothing in the "generate_esdf_srv" service. So can you help me with how to get esdf map from tsdf map?
Thanks sincerely!

Austen-G, 2024-03-01:
I was able to do this by saving the output .tsdf file as a .vxblx file, calling the load_map service with this file, and as long as a static transform is defined between /map and /sensor_frame somewhere it worked:

[image]

12ahmed52, 2024-04-29:
i cannot generate the esdf map although i called the /save_esdf map service and when i load the generated map is the vox_blox planner it says that the map is tsdf not esdf chow can I generate the esdf map

Austen-G, 2024-04-29:
I haven't tried those, I just save the topic outputs into .bag files and play them back when I need them as input. 

Also I'm not a dev on this or anything, I've just been tinkering with it to support some research.  But something I've found since my last comment:

For my purposes at least, voxblox ground truth doesn't let me mess with the specific settings (or at least I don't know how to) that help make the maps more usable (truncation distance, etc), so I use a simulated sensor in Gazebo instead. If positioned far enough away you can view the whole map, but in reality putting that sensor into a simulated slam package like ORB-Slam can likely be used to help move it around.

You can do some movement without SLAM, but it's difficult to completely generate the map with a sensor without it.

Wshtm, 2024-05-10:
@Austen-G Hi, could you please tell me that how to complete "the static transform between /map and /sensor_frame"? I try to use voxblox/load_map to load the map generated by Voxblox_gound_truth, but it returns error. Thank you so much!

Austen-G, 2024-05-10:
A couple of ways, to keep things easy when I use simulated sensors in Gazebo, I use a plugin to publish the exact sensor position/rotation as a transform. The localization portion of a SLAM solution should also be able to publish this transform.

Since you're not building maps yourself though, try just using the static_transform_publisher from the tf package