ethz-asl/voxblox #389: catkin build error about grpc

bulinx, 2021-12-17:
Hello!
Thanks for this awesome framework!
In fact, I successfully used voxblox before. I don’t know if it’s because I updated the source. When I ### roslaunch again recently, an error was reported：

`/home/ok/catkin_voxgraph/devel/lib/voxblox_ros/tsdf_server: error while loading shared libraries: libeigen_conversions.so: cannot open shared object file: No such file or directory
`
Because I couldn’t solve it many times, I chose to rebuild the project.an error was reported：

`$ catkin build voxblox_ros
--------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/melodic
Workspace:                   /home/ok/catkin_vox
--------------------------------------------------------
Build Space:        [exists] /home/ok/catkin_vox/build
Devel Space:        [exists] /home/ok/catkin_vox/devel
Install Space:      [unused] /home/ok/catkin_vox/install
Log Space:         [missing] /home/ok/catkin_vox/logs
Source Space:       [exists] /home/ok/catkin_vox/src
DESTDIR:            [unused] None
--------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
--------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------
[build] Found '17' packages in 0.0 seconds.                                                                    
[build] Updating package table.                                                                                
Starting  >>> catkin_simple                                                                                    
Starting  >>> grpc                                                                                             
Finished  <<< catkin_simple                       [ 3.6 seconds ]                                              
Starting  >>> eigen_catkin                                                                                     
Starting  >>> gflags_catkin                                                                                    
Starting  >>> voxblox_msgs                                                                                     
Finished  <<< eigen_catkin                        [ 3.3 seconds ]                                              
Finished  <<< voxblox_msgs                        [ 5.1 seconds ]                                              
_______________________________________________________________________________________________________________
Errors     << grpc:make /home/ok/catkin_vox/logs/grpc/build.make.000.log                                       
Cloning into 'grpc_build'...
Note: checking out 'v1.34.1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at e5276ec36 Merge pull request #25135 from yulin-liang/fix_macos_distribtest
Submodule 'third_party/abseil-cpp' (https://github.com/abseil/abseil-cpp.git) registered for path 'third_party/abseil-cpp'
Submodule 'third_party/benchmark' (https://github.com/google/benchmark) registered for path 'third_party/benchmark'
Submodule 'third_party/bloaty' (https://github.com/google/bloaty.git) registered for path 'third_party/bloaty'
Submodule 'third_party/boringssl-with-bazel' (https://github.com/google/boringssl.git) registered for path 'third_party/boringssl-with-bazel'
Submodule 'third_party/cares/cares' (https://github.com/c-ares/c-ares.git) registered for path 'third_party/cares/cares'
Submodule 'third_party/envoy-api' (https://github.com/envoyproxy/data-plane-api.git) registered for path 'third_party/envoy-api'
Submodule 'third_party/googleapis' (https://github.com/googleapis/googleapis.git) registered for path 'third_party/googleapis'
Submodule 'third_party/googletest' (https://github.com/google/googletest.git) registered for path 'third_party/googletest'
Submodule 'third_party/libuv' (https://github.com/libuv/libuv.git) registered for path 'third_party/libuv'
Submodule 'third_party/protobuf' (https://github.com/google/protobuf.git) registered for path 'third_party/protobuf'
[... 39 lines cut ...]
make[2]: *** [grpc/src/build_grpc-stamp/build_grpc-download] Error 1
make[1]: *** [CMakeFiles/build_grpc.dir/all] Error 2
make: *** [all] Error 2
cd /home/ok/catkin_vox/build/grpc; catkin build --get-env grpc | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................................................
Failed     << grpc:make                           [ Exited with code 2 ]                                       
Failed    <<< grpc                                [ 9 minutes and 36.3 seconds ]                               
Abandoned <<< glog_catkin                         [ Unrelated job failed ]                                     
Abandoned <<< eigen_checks                        [ Unrelated job failed ]                                     
Abandoned <<< minkindr                            [ Unrelated job failed ]                                     
Abandoned <<< minkindr_conversions                [ Unrelated job failed ]                                     
Abandoned <<< protobuf_catkin                     [ Unrelated job failed ]                                     
Abandoned <<< voxblox                             [ Unrelated job failed ]                                     
Abandoned <<< voxblox_rviz_plugin                 [ Unrelated job failed ]                                     
Abandoned <<< voxblox_ros                         [ Unrelated job failed ]                                     
^C[build - 10:13.7] [4/13 complete] [1/12 jobs] [0 queued] [1 failed] [gflags_catkin:make (25%) - 10:10.1]      [build] Interrupted by user!                                                                                   
[build] Summary: 3 of 13 packages succeeded.                                                                   
[build]   Ignored:   4 packages were skipped or are blacklisted.                                               
[build]   Warnings:  None.                                                                                     
[build]   Abandoned: 9 packages were abandoned.                                                                
[build]   Failed:    1 packages failed.                                                                        
[build] Runtime: 10 minutes and 13.8 seconds total.                                  `

I tried many times and it should not be a problem with the network. At the same time, an SSH connection is established.
Hope you can give me some guidance, thank you very much！

bulinx, 2021-12-17:
> Hello! Thanks for this awesome framework! In fact, I successfully used voxblox before. I don’t know if it’s because I updated the source. When I ** _roslaunch_** again recently, an error was reported：
> 
> ` /catkin_voxgraph/devel/lib/voxblox_ros/tsdf_server: error while loading shared libraries: libeigen_conversions.so: cannot open shared object file: No such file or directory`
> 
> Because I couldn’t solve it many times, I chose to rebuild the project.an error was reported：
> 
> ```
> $ catkin build voxblox_ros
> - --------------------------------------------------------
> Profile:                     default
> Extending:        [explicit] /opt/ros/melodic
> Workspace:                   /home/ok/catkin_vox
> --------------------------------------------------------
> Build Space:        [exists] /home/ok/catkin_vox/build
> Devel Space:        [exists] /home/ok/catkin_vox/devel
> Install Space:      [unused] /home/ok/catkin_vox/install
> Log Space:         [missing] /home/ok/catkin_vox/logs
> Source Space:       [exists] /home/ok/catkin_vox/src
> DESTDIR:            [unused] None
> --------------------------------------------------------
> Devel Space Layout:          merged
> Install Space Layout:        None
> --------------------------------------------------------
> Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
> Additional Make Args:        None
> Additional catkin Make Args: None
> Internal Make Job Server:    True
> Cache Job Environments:      False
> --------------------------------------------------------
> Whitelisted Packages:        None
> Blacklisted Packages:        None
> --------------------------------------------------------
> Workspace configuration appears valid.
> --------------------------------------------------------
> [build] Found '17' packages in 0.0 seconds.                                                                    
> [build] Updating package table.                                                                                
> Starting  >>> catkin_simple                                                                                    
> Starting  >>> grpc                                                                                             
> Finished  <<< catkin_simple                       [ 3.6 seconds ]                                              
> Starting  >>> eigen_catkin                                                                                     
> Starting  >>> gflags_catkin                                                                                    
> Starting  >>> voxblox_msgs                                                                                     
> Finished  <<< eigen_catkin                        [ 3.3 seconds ]                                              
> Finished  <<< voxblox_msgs                        [ 5.1 seconds ]                                              
> _______________________________________________________________________________________________________________
> Errors     << grpc:make /home/ok/catkin_vox/logs/grpc/build.make.000.log                                       
> Cloning into 'grpc_build'...
> Note: checking out 'v1.34.1'.
> 
> You are in 'detached HEAD' state. You can look around, make experimental
> changes and commit them, and you can discard any commits you make in this
> state without impacting any branches by performing another checkout.
> 
> If you want to create a new branch to retain commits you create, you may
> do so (now or later) by using -b with the checkout command again. Example:
> 
>   git checkout -b <new-branch-name>
> 
> HEAD is now at e5276ec36 Merge pull request #25135 from yulin-liang/fix_macos_distribtest
> Submodule 'third_party/abseil-cpp' (https://github.com/abseil/abseil-cpp.git) registered for path 'third_party/abseil-cpp'
> Submodule 'third_party/benchmark' (https://github.com/google/benchmark) registered for path 'third_party/benchmark'
> Submodule 'third_party/bloaty' (https://github.com/google/bloaty.git) registered for path 'third_party/bloaty'
> Submodule 'third_party/boringssl-with-bazel' (https://github.com/google/boringssl.git) registered for path 'third_party/boringssl-with-bazel'
> Submodule 'third_party/cares/cares' (https://github.com/c-ares/c-ares.git) registered for path 'third_party/cares/cares'
> Submodule 'third_party/envoy-api' (https://github.com/envoyproxy/data-plane-api.git) registered for path 'third_party/envoy-api'
> Submodule 'third_party/googleapis' (https://github.com/googleapis/googleapis.git) registered for path 'third_party/googleapis'
> Submodule 'third_party/googletest' (https://github.com/google/googletest.git) registered for path 'third_party/googletest'
> Submodule 'third_party/libuv' (https://github.com/libuv/libuv.git) registered for path 'third_party/libuv'
> Submodule 'third_party/protobuf' (https://github.com/google/protobuf.git) registered for path 'third_party/protobuf'
> Submodule 'third_party/protoc-gen-validate' (https://github.com/envoyproxy/protoc-gen-validate.git) registered for path 'third_party/protoc-gen-validate'
[... 39 lines cut ...]
> make[2]: *** [grpc/src/build_grpc-stamp/build_grpc-download] Error 1
> make[1]: *** [CMakeFiles/build_grpc.dir/all] Error 2
> make: *** [all] Error 2
> cd /home/ok/catkin_vox/build/grpc; catkin build --get-env grpc | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
[... cut ...]
> [build]   Warnings:  None.                                                                                     
> [build]   Abandoned: 9 packages were abandoned.                                                                
> [build]   Failed:    1 packages failed.                                                                        
> [build] Runtime: 10 minutes and 13.8 seconds total.        
> ```
> 
> I tried many times and it should not be a problem with the network. At the same time, an SSH connection is established. Hope you can give me some guidance, thank you very much！

bulinx, 2021-12-17:
sudo apt-get install ros-melodic-eigen-conversions

closed 2021-12-17 by bulinx

city945, 2022-04-08:
> sudo apt-get install ros-melodic-eigen-conversions

Hi,I meet the same problem. And this doesn't work,have you  solved it? Thank you in advance.

city945, 2022-04-08:
I solved it by `sudo apt install -y ros-melodic-grpc`，https://github.com/CogRob/catkin_grpc/issues/34