ethz-asl/voxblox #390: voxblox_ros build error

Kwyper, 2022-01-25:
Hi I'm a newcomer to this. I've installed all the missing dependencies, yet I still can't build. Does anyone have any idea? Thanks in advance!
```Profile:                     default
Extending:        [explicit] /opt/ros/kinetic
Workspace:                   /home/wy/voxblox
-------------------------------------------------------
Build Space:        [exists] /home/wy/voxblox/build
Devel Space:        [exists] /home/wy/voxblox/devel
Install Space:      [unused] /home/wy/voxblox/install
Log Space:          [exists] /home/wy/voxblox/logs
Source Space:       [exists] /home/wy/voxblox/src
DESTDIR:            [unused] None
-------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
-------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------
[build] Found '17' packages in 0.0 seconds.                                    
[build] Package table is up to date.                                           
Starting  >>> catkin_simple                                                    
Starting  >>> grpc                                                             
Finished  <<< catkin_simple                       [ 0.4 seconds ]              
Starting  >>> eigen_catkin                                                     
Starting  >>> gflags_catkin                                                    
Finished  <<< eigen_catkin                        [ 0.4 seconds ]              
Starting  >>> voxblox_msgs                                                     
Finished  <<< gflags_catkin                       [ 0.3 seconds ]              
Starting  >>> glog_catkin                                                      
Finished  <<< glog_catkin                         [ 0.4 seconds ]              
Starting  >>> eigen_checks                                                     
Finished  <<< eigen_checks                        [ 0.2 seconds ]              
Starting  >>> minkindr                                                         
Finished  <<< voxblox_msgs                        [ 1.8 seconds ]              
Finished  <<< minkindr                            [ 0.5 seconds ]              
Starting  >>> minkindr_conversions                                             
Finished  <<< minkindr_conversions                [ 0.5 seconds ]              
Finished  <<< grpc                                [ 6.7 seconds ]              
Starting  >>> protobuf_catkin                                                  
Finished  <<< protobuf_catkin                     [ 0.2 seconds ]              
Starting  >>> voxblox                                                          
Finished  <<< voxblox                             [ 1 minute and 16.4 seconds ]
Starting  >>> voxblox_rviz_plugin                                              
Finished  <<< voxblox_rviz_plugin                 [ 4.4 seconds ]              
Starting  >>> voxblox_ros                                                      
_______________________________________________________________________________
Errors     << voxblox_ros:make /home/wy/voxblox/logs/voxblox_ros/build.make.001.log
In file included from /home/wy/voxblox/src/voxblox/voxblox/include/voxblox/core/esdf_map.h:8:0,
                 from /home/wy/voxblox/src/voxblox/voxblox_ros/include/voxblox_ros/simulation_server.h:9,
                 from /home/wy/voxblox/src/voxblox/voxblox_ros/src/simulation_server.cc:1:
/home/wy/voxblox/src/voxblox/voxblox_ros/src/simulation_server.cc: In member function ‘bool voxblox::SimulationServer::generatePlausibleViewpoint(voxblox::FloatingPoint, voxblox::Point*, voxblox::Point*) const’:
/home/wy/voxblox/src/voxblox/voxblox_ros/src/simulation_server.cc:165:3: error: no matching function for call to ‘CheckNotNull(const char [66], int, const char [26], const std::unique_ptr<voxblox::SimulationWorld>&)’
   CHECK_NOTNULL(world_);
   ^
In file included from /home/wy/voxblox/src/voxblox/voxblox/include/voxblox/core/esdf_map.h:8:0,
                 from /home/wy/voxblox/src/voxblox/voxblox_ros/include/voxblox_ros/simulation_server.h:9,
                 from /home/wy/voxblox/src/voxblox/voxblox_ros/src/simulation_server.cc:1:
/usr/include/glog/logging.h:1286:4: note: candidate: template<class T> T* google::CheckNotNull(const char*, int, const char*, T*)
 T* CheckNotNull(const char *file, int line, const char *names, T* t) {
    ^
/usr/include/glog/logging.h:1286:4: note:   template argument deduction/substitution failed:
In file included from /home/wy/voxblox/src/voxblox/voxblox/include/voxblox/core/esdf_map.h:8:0,
                 from /home/wy/voxblox/src/voxblox/voxblox_ros/include/voxblox_ros/simulation_server.h:9,
                 from /home/wy/voxblox/src/voxblox/voxblox_ros/src/simulation_server.cc:1:
/home/wy/voxblox/src/voxblox/voxblox_ros/src/simulation_server.cc:165:3: note:   mismatched types ‘T*’ and ‘std::unique_ptr<voxblox::SimulationWorld>’
   CHECK_NOTNULL(world_);
   ^
/home/wy/voxblox/src/voxblox/voxblox_ros/src/simulation_server.cc: In member function ‘void voxblox::SimulationServer::generateSDF()’:
/home/wy/voxblox/src/voxblox/voxblox_ros/src/simulation_server.cc:221:5: error: no matching function for call to ‘CheckNotNull(const char [66], int, const char [26], std::unique_ptr<voxblox::SimulationWorld>&)’
     CHECK_NOTNULL(world_);
     ^
In file included from /home/wy/voxblox/src/voxblox/voxblox/include/voxblox/core/esdf_map.h:8:0,
                 from /home/wy/voxblox/src/voxblox/voxblox_ros/include/voxblox_ros/simulation_server.h:9,
                 from /home/wy/voxblox/src/voxblox/voxblox_ros/src/simulation_server.cc:1:
/usr/include/glog/logging.h:1286:4: note: candidate: template<class T> T* google::CheckNotNull(const char*, int, const char*, T*)
 T* CheckNotNull(const char *file, int line, const char *names, T* t) {
    ^
/usr/include/glog/logging.h:1286:4: note:   template argument deduction/substitution failed:
In file included from /home/wy/voxblox/src/voxblox/voxblox/include/voxblox/core/esdf_map.h:8:0,
                 from /home/wy/voxblox/src/voxblox/voxblox_ros/include/voxblox_ros/simulation_server.h:9,
                 from /home/wy/voxblox/src/voxblox/voxblox_ros/src/simulation_server.cc:1:
/home/wy/voxblox/src/voxblox/voxblox_ros/src/simulation_server.cc:221:5: note:   mismatched types ‘T*’ and ‘std::unique_ptr<voxblox::SimulationWorld>’
     CHECK_NOTNULL(world_);
     ^
make[2]: *** [CMakeFiles/voxblox_ros.dir/src/simulation_server.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/voxblox_ros.dir/all] Error 2
make: *** [all] Error 2
cd /home/wy/voxblox/build/voxblox_ros; catkin build --get-env voxblox_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << voxblox_ros:make                    [ Exited with code 2 ]       
Failed    <<< voxblox_ros                         [ 1 minute and 40.7 seconds ]
[build] Summary: 12 of 13 packages succeeded.                                  
[build]   Ignored:   4 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 3 minutes and 9.3 seconds total. ```

Kwyper, 2022-01-31:
Solved by installing glog and gflags and then re-installing the packages with SSH instead of https.

closed 2022-01-31 by Kwyper

Southyang, 2023-03-10:
`sudo apt-get install libgoogle-glog-dev libgflags-dev`
Error after installation：
[image]

wangjinhoon, 2024-03-01:
How do I solve this problem?