ethz-asl/voxblox #408: Possible memory alignment problem

salihmarangoz, 2023-03-17:
I have copied [tsdf_server.cc](https://github.com/ethz-asl/voxblox/blob/master/voxblox_ros/src/tsdf_server.cc) and modified it to my needs. But I am getting memory alignment related segmentation faults (using AddressSanitizer) in this line: https://github.com/ethz-asl/voxblox/blob/master/voxblox/src/integrator/tsdf_integrator.cc#L571

It looked weird to me because there is no Eigen-related data structure in ThreadSafeIndex. When I removed `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` here: https://github.com/ethz-asl/voxblox/blob/master/voxblox/include/voxblox/integrator/integrator_utils.h#L24 the problem was solved. Do you think it is ok to use `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` here? What is the purpose? Or, is it a false positive with AddressSanitizer?

dikokob, 2024-03-26:
Did you ever find a solution to the problem?

salihmarangoz, 2024-03-26:
> Did you ever find a solution to the problem?
> 

I think the problem is related to Eigen being a header-only library and being compiled with different options with the linked libraries such as PCL. 

To overcome this problem, Voxblox and its dependencies that includes Eigen must be compiled with the same compile options. So, the memory alignment will be the same for everything.