ethz-asl/voxblox #412: failed making grpc

AreteQin, 2023-12-23:
Hi

Thank you for this excellent work.

I encountered an error while compiling using SSH and catkin build.

Here is the full log:

```Bash
qin@qin-desktop:~/catkin_ws$ catkin build voxblox_ros
--------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/noetic
Workspace:                   /home/qin/catkin_ws
--------------------------------------------------------
Build Space:        [exists] /home/qin/catkin_ws/build
Devel Space:        [exists] /home/qin/catkin_ws/devel
Install Space:      [unused] /home/qin/catkin_ws/install
Log Space:          [exists] /home/qin/catkin_ws/logs
Source Space:       [exists] /home/qin/catkin_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
--------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------
[build] Found 21 packages in 0.0 seconds.                                                                                              
[build] Package table is up to date.                                                                                                   
Starting  >>> catkin_simple                                                                                                            
Starting  >>> grpc                                                                                                                     
Finished  <<< catkin_simple                       [ 0.1 seconds ]                                                                      
Starting  >>> eigen_catkin                                                                                                             
Starting  >>> gflags_catkin                                                                                                            
Starting  >>> voxblox_msgs                                                                                                             
Finished  <<< gflags_catkin                       [ 0.1 seconds ]                                                                      
Starting  >>> glog_catkin                                                                                                              
Finished  <<< eigen_catkin                        [ 0.1 seconds ]                                                                      
Finished  <<< glog_catkin                         [ 0.1 seconds ]                                                                      
Starting  >>> eigen_checks                                                                                                             
Finished  <<< voxblox_msgs                        [ 0.2 seconds ]                                                                      
Finished  <<< eigen_checks                        [ 0.1 seconds ]                                                                      
Starting  >>> minkindr                                                                                                                 
Finished  <<< minkindr                            [ 0.1 seconds ]                                                                      
Starting  >>> minkindr_conversions                                                                                                     
Finished  <<< minkindr_conversions                [ 0.1 seconds ]                                                                      
_______________________________________________________________________________________________________________________________________
Errors     << grpc:make /home/qin/catkin_ws/logs/grpc/build.make.010.log                                                               
Cloning into 'grpc_build'...
Note: switching to 'v1.34.1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:
[... 91 lines cut ...]
Cloning into '/home/qin/catkin_ws/build/grpc/grpc_build/third_party/boringssl-with-bazel'...
CMake Error at grpc/tmp/build_grpc-gitclone.cmake:52 (message):
  Failed to update submodules in: '/home/qin/catkin_ws/build/grpc/grpc_build'

make[2]: *** [CMakeFiles/build_grpc.dir/build.make:91: grpc/src/build_grpc-stamp/build_grpc-download] Error 1
make[1]: *** [CMakeFiles/Makefile2:276: CMakeFiles/build_grpc.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/qin/catkin_ws/build/grpc; catkin build --get-env grpc | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

.......................................................................................................................................
Failed     << grpc:make                           [ Exited with code 2 ]                                                               
Failed    <<< grpc                                [ 11 minutes and 16.2 seconds ]                                                      
Abandoned <<< protobuf_catkin                     [ Unrelated job failed ]                                                             
Abandoned <<< voxblox                             [ Unrelated job failed ]                                                             
Abandoned <<< voxblox_rviz_plugin                 [ Unrelated job failed ]                                                             
Abandoned <<< voxblox_ros                         [ Unrelated job failed ]                                                             
[build] Summary: 8 of 13 packages succeeded.                                                                                           
[build]   Ignored:   8 packages were skipped or are skiplisted.                                                                        
[build]   Warnings:  None.                                                                                                             
[build]   Abandoned: 4 packages were abandoned.                                                                                        
[build]   Failed:    1 packages failed.                                                                                                
[build] Runtime: 11 minutes and 16.3 seconds total.
```

Thank you for any help in advance. I am sorry if I missed sth already there.

kirilllzaitsev, 2024-01-02:
@AreteQin hi, facing the same issue, have you achieved any progress?

AreteQin, 2024-01-02:
It's the problem of the internet for me. Using the hotspot of the phone solved the problem.